demiangomez/geode #21: etm and ts/etm plotting

smalleyr, 2017-10-31:
Peter and I were looking at the ETM and ts/etm plotting today.

We found out why most of the ones not plotting were "failing" - they had less than 4 observations - some had between 3 and 1, and at least one had 0 observations, no rinex file in the archive, and no return from postgresql in the station list).

Here is our request. Separate the ETM calculations and plotting. The ETM class should calculate the ETM and provide an object to pass around in the program or write it to disk (with time series optional as it is now). Put all zeros in the etm parameters result to signify there was no fit. This will allow the objet/file to be used to pass just the time series.

In the PlotETM class, plot the time series (from an object within the program or from a file) as the basic result, and plot the etm if the parameters indicate an ETM was found (at least one amplitude not zero), or no ETM if it was not found. Should be able to plot just the time series if ETM exists. Should also be able to plot the residuals after removing the ETM (to see if there is any structure in the residuals).

There may be some complications if one want's to do the jumps (know where they are), but probably best for just time series to plot raw time series.

demiangomez, 2017-12-31:
This was partially implemented. I will work on the rest during the next months.

demiangomez, 2018-06-30:
Added an option -r to plot residuals. Calculation separated from plot. I'm closing this issue.

closed 2018-06-30 by demiangomez