demiangomez/geode #27: example - bad rinex our processing, nrcan gets good soln

smalleyr, 2017-12-30:
error message from ArchiveService run, there should be a pdf attached with NRCAN solution.

says problem is with sampling interval.

RINEX sampling interval could not be determined. The output from RinSum was:
# RinSum, part of the GPS Toolkit, Ver 2.2 10/31/13, Run 2017/12/30 00:39:50
+++++++++++++ RinSum summary of Rinex obs file production/rinex/6e271835-0489-4a0f-9e9a-dde6b5be02d3/lo101310.00o +++++++++++++
Warning : Failed to read header: text 0:Unidentified label: >ANTENNA: DEL<
text 1:In record 0
text 2:In file production/rinex/6e271835-0489-4a0f-9e9a-dde6b5be02d3/lo101310.00o
text 3:Near file line 12
location 0:/Volumes/UsersDrive/moved_user/smalley/Downloads/gpstk-2.5.src/dev/ext/lib/FileHandling/RINEX3/Rinex3ObsHeader.cpp:1425
location 1:/Volumes/UsersDrive/moved_user/smalley/Downloads/gpstk-2.5.src/dev/ext/lib/FileHandling/RINEX3/Rinex3ObsHeader.cpp:1471
location 2:/Volumes/UsersDrive/moved_user/smalley/Downloads/gpstk-2.5.src/dev/ext/lib/FileHandling/FFStream.cpp:150
location 3:/Volumes/UsersDrive/moved_user/smalley/Downloads/gpstk-2.5.src/dev/ext/lib/FileHandling/FFStream.hpp:184
location 4:/Volumes/UsersDrive/moved_user/smalley/Downloads/gpstk-2.5.src/dev/ext/lib/FileHandling/FFStream.hpp:184

 Header dump follows.
---------------------------------- REQUIRED ----------------------------------
Rinex Version  2.00,  File type O    BSERVATION DATA,  System G (GPS).
Prgm: ASHTORIN,  Run: 31 - MAY - 00 16:20,  By: 
Marker type: .
Observer : AO_,  Agency: 
Rec#: GN-1331,  Type: TOPCON GP-R1DY,  Vers: 
Antenna # : BX-3191,  Type : 
Position      (XYZ,m) : (1911998.2600, -4237323.8800, -4352388.4900).
Antenna Delta (HEN,m) : (0.0000, 0.0000, 0.0000).
Time of first obs -002/12/01 00:00:00.000 UNK
(This header is VALID)
---------------------------------- OPTIONAL ----------------------------------
Marker number : 
Comments (3) :

Concatenated from 2 rinex files by RNXCAT on 20 Jul 00

-------------------------------- END OF HEADER --------------------------------
RinSum timing: processing 0.007 sec, wallclock: 0 sec.

/lo101310.00d.Z: (file moved to /Volumes/UsersDrive/repository/data_rejected/bad_rinex/2000/131)
[lo101310.pdf](https://github.com/demiangomez/Parallel.GAMIT/files/1594587/lo101310.pdf)

demiangomez, 2017-12-31:
If you read the message output, you'll see that the problem is not the interval. The problem is 
`Warning : Failed to read header: text 0:Unidentified label: >ANTENNA: DEL<`
Therefore, the RINEX header seems to be corrupt. Please upload the RINEX file with the problem. You can attach files to this interface. Let's keep this thread until we figure out what is going on.

smalleyr, 2017-12-31:
OK, sorry, I only read the first line of the error message "RINEX sampling interval could not be determined. " and since it had a sample time problem I thought that was it.

To make a long story short - I found the antenna type in the OSU archive station info. But there were field name errors in the header in the rinex file. It was still falling over on some non-descript error reading the header after I fixed the obvious errors.

So I passed it through teqc into a new file name (1 instead of 0) after which it processed.

This is a posgar site (or 2, LO10, measured 1995/2000, and L10B, measured 2006) by Golfo San Jorge. I'm going to drop it for my processing. Is it worth fixing?

smalleyr, 2017-12-31:
You can close it. See comments in the github issues thread on it.

> If you read the message output, you'll see that the problem is not the
> interval. The problem is
> |Warning : Failed to read header: text 0:Unidentified label: >ANTENNA:
> DEL<|
> Therefore, the RINEX [...]

-- 
Dr. Robert Smalley, Jr.
Research Professor
Center for Earthquake Research and Information
The University of Memphis
3890 Central Ave.
Memphis, TN 38152-3050
901-678-4929
901-678-4734 fax
rsmalley@memphis.edu
http://www.ceri.memphis.edu
office: 3892 Central, Room 103

closed 2018-01-01 by demiangomez