demiangomez/geode #29: ppp reporting no station info

smalleyr, 2017-12-31:
ppp reports no station info found when database says there is station info. problem arises when station info start/end is to second, or within a second of rinex start end.

demiangomez, 2017-12-31:
Please check that you are running the latest version of the code.

demiangomez, 2018-01-20:
Found that the problem is in pyScanArchive::hash_check. The station info object is invoked as `stninfo = pyStationInfo.StationInfo(cnn, soln['NetworkCode'], soln['StationCode'], pyDate.Date(year=soln['Year'],doy=soln['DOY']))` but should include the time as well. This information is not available in the ppp_soln table so it might be necessary to include a join between ppp_soln and rinex.

demiangomez, 2018-06-30:
We need to check all the code for this inconsistency. I think all problems were solved, but we should double check.

demiangomez, 2019-03-13:
Mostly solved after new pyStationInfo module.

closed 2019-03-13 by demiangomez