demiangomez/geode #31: database organizational change

smalleyr, 2018-01-14:
Change in the "NetworkCode" to make it more useful for multiple applications of database.

Make "NetworkCode" meaningless to the end user. There could be a base network code, say n00, where all new sites with unique names go. Duplicate names go into networks n01, n02, etc. as such sites come into the database.

Second network code where user can organize  groups of sites based on need or preference.

Example - IGN and CAP use same underlying database but can organize the interface how they want/need (IGN can have IGS, SIR, RMS, etc. networks, while CAP can have ARG, ARS, CHI, CHS, PIF, MAU, PIS, ...)

So  any given site could be in multiple "networks". Need to be careful when deleting sites - if in multiple application networks, does not get touched at the nXX level. If a unique site sends a message. Or always send message about sites affiniies.

demiangomez, 2018-01-18:
This can be done but not exactly as suggested. User can make the NetworkCode follow the suggested pattern if desired. Another table can contain logical divisions of stations, but this wouldn't be a field in the "stations" table.

demiangomez, 2024-05-10:
This has been partially implemented with the ISO3 country code needed for RINEX3. New modules country_converter and geopy are needed. Marking as resoved.

closed 2024-05-10 by demiangomez