demiangomez/geode #57: Two format dates

Salacas, 2024-06-18:
Users should be able to set date in two formats: gregorian and day of year, when updating or inserting station-info.

francolau, 2024-07-01:
Date format types (needed check DOY checkbox to type in doy type format)
[image]

demiangomez, 2024-07-05:
DOY format should also take time into account!

[image]

demiangomez, 2024-07-05:
When switching between the two formats, time is removed and ignored

[image]

commit c5bb8749 (2024-07-01): closed #57,  closed #58, closed #59,  closed #61 filters, map rendering and station info dates