demiangomez/geode #60: Search by station code

Salacas, 2024-06-18:
On the main page, when the search by station code field is left empty after writing to it the stations are still filtered, unless Enter is pressed.

demiangomez, 2024-08-08:
There is also a strange behavior when you type a station code, see video below:

[Screencast from 08-08-2024 10:21:08 AM.webm](https://github.com/user-attachments/assets/ece9de1a-532c-4543-b50a-378e60a29423)

demiangomez, 2024-08-08:
Here is a longer version showing that this always happens. Also, the "code field is left empty after writing to it the stations are still filtered, unless Enter is pressed" seems to still be a problem with the addition that the map move to what it looks like a random position.

[Screencast from 08-08-2024 10:24:25 AM.webm](https://github.com/user-attachments/assets/07ce4969-8a62-4c29-bbe9-f81d641af0fb)

commit c1d41836 (2024-08-09): closed #60, closed #66, closed #84
closed 2024-08-12 by francolau