demiangomez/geode #62: Side list

Salacas, 2024-06-18:
On the main page, display a list on the left side of the map. This list should contain all stations displayed on the map. The data for each station in the list consists of only a few columns, which will be determined in the future.

demiangomez, 2024-06-19:
I'm including here an example from the IGS websire
[image]

Salacas, 2024-08-06:
[image]

Salacas, 2024-08-06:
the commit 1e4a50f6b4620a8ad6189dae2bfc9aff5749f9ba solved the issue
[image]

closed 2024-08-06 by Salacas