demiangomez/geode #63: Change station icon based on status

Salacas, 2024-06-18:
Inside the map, show each station with an alert sign if there is an issue with it. Otherwise, the icon should be a green triangle.

Backend should implement its version of StationInfoRecord.station_info_gaps() inside [https://github.com/demiangomez/Parallel.GAMIT/blob/master/classes/pyStationInfo.py](url) in order to set the station status

demiangomez, 2024-06-19:
We can probably use the same standard icons has in PG:
http://maps.google.com/mapfiles/kml/shapes/placemark_square.png -> in green or red if we are showing continuous vs survey data (determined by the number of observations: < 100 survey; > 100 continuous).
http://maps.google.com/mapfiles/kml/shapes/caution.png -> for a problem with metadata

demiangomez, 2024-07-31:
Related to this modification, we need to display the type of problem somewhere. Right now, the icon shows up but there is not indication on what the issue is. Maybe upon navigating to station, a message under the station name should appear showing the type of problem. There are only four possible messages -> See generate_kml_stninfo in https://github.com/demiangomez/Parallel.GAMIT/blob/fbe6c579d0caa9242da435d97727aaf1bd8184e6/parallel_gamit/GenerateKml.py#L267

demiangomez, 2024-07-31:
This information mentioned above could also be shown in the popup for the station:
[image]

demiangomez, 2024-08-02:
Instead of the word "STATUS" maybe we can show the station type: Continuous or Campaign. If not determined, then maybe "Station type not defined". This issue is somewhat related to #76

Salacas, 2024-08-08:
> Instead of the word "STATUS" maybe we can show the station type: Continuous or Campaign. If not determined, then maybe "Station type not defined". This issue is somewhat related to #76

done

[image]

closed 2024-08-19 by francolau