demiangomez/geode #77: Submit a record with empty date end is filled automatically

marafigue, 2024-08-05:
When you submit an Information record with empty "DATE END", it automatically fills the date with some date in the future:

[image]

[image]

demiangomez, 2024-08-05:
This is really dangerous! @marafigue please make sure that all of the station info that you created with this tool have the appropriate date end. If needed, make them NULL using pgAdmin. @Salacas the behavior here should be to allow this field to be NULL. All processes within Parallel.GAMIT assume that a NULL date end means that this station information record is OPEN and, therefore, all new data should have this record applied. Although writing a date in the future accomplishes the same thing, I am not entirely sure what could happen to all the processes that expect to see NULL here.

demiangomez, 2024-08-06:
Problem with date persists and now shows other issues:

NULL date represented as 1970. This example is from RDCM
[image]

Date in DOY format and date picker is not synchronized
[image]
[image]

If you edit a field but escape the window (using the ESC key) the change appears to be applied if you open the edit window again. This actually applies to ALL fields, not just the date. When you click on the X to close, then the information is discarded, or if you load a different record.

francolau, 2024-08-08:
> If you edit a field but escape the window (using the ESC key) the change appears to be applied if you open the edit window again. This actually applies to ALL fields, not just the date. When you click on the X to close, then the information is discarded, or if you load a different record.

ESC Key never save the information, but we also check why the last value is hold on the form.

francolau, 2024-08-08:
> ESC Key never save the information, but we also check why the last value is hold on the form.

Now ESC Key act like cross key on header.

commit d51a45e5 (2024-08-07): closed #64, closed #75, closed #77, closed #79, closed #81 Station info fixes
commit c1e38f5d (2024-08-08): closed #77, Escape key on station info modal handled and dates formatted
closed 2024-08-08 by francolau