demiangomez/geode #82: Improve visualization of two or more stations that are too close or have the same coordinates

marafigue, 2024-08-07:
When two stations have the same coordinates (or are close enough, depending on the map scale), their markers are overlapped and it is impossible to see both

[image]
[image]

We might need to implement something like this in Google Earth:
[image]

demiangomez, 2025-01-15:
[14:00, 1/15/2025] Demian Gomez: si pueden de alguna forma obtener la escala del mapa (eg. 1:30.000 o 1:100.000) de acuerdo al zoom, la mejor manera es usar lo que en cartografía se llamar la regla de la agudeza visual
[14:01, 1/15/2025] Demian Gomez: la regla dice que no es posible distinguir dos elementos (puntos) en un mapa si la distancia entre ellos es menor que 0.2 [mm] * ESCALA
[14:02, 1/15/2025] Demian Gomez: entonces, si la escala es 1:10.000, la distancia mínima que se puede resolver es 0.2 * 10.000 = 2.000 mm = 2 metros
[14:04, 1/15/2025] Demian Gomez: entonces, si el mapa está en escala 1:100.000 y la distancia entre dos estaciones es 15 metros, como la agudeza visual es 20 m para 1:100.000, hay que desplegar alguna opción para que el usuario puede saber que hay dos (o más) estaciones
[14:05, 1/15/2025] Demian Gomez: en nuestro caso, como los puntos de las estaciones son grandes, muy posiblemente tengan que aumentar ese factor de 0.2 a otra cosa, habrá que probar

francolau, 2025-04-03:
Done

demiangomez, 2025-04-03:
Not sure if it was the intention, but this appears to only work at some zoom level. 

[image]

[image]

demiangomez, 2025-06-11:
Done

closed 2025-06-11 by demiangomez