demiangomez/geode #85: Icon not reflecting station's state

demiangomez, 2024-08-08:
Some stations have stationinfo problems, as reported by pyStationInfo

[image]

or using the StationInfo object directly:

(venv) demian@demian-dell:~/Parallel.GAMIT$ python3
Python 3.10.12 (main, Jul 29 2024, 16:56:48) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyStationInfo
>>> import dbConnection
>>> cnn = dbConnection.Cnn('gnss_data.cfg')
>>> stn=pyStationInfo.StationInfo(cnn, 'sag', 'baqu')
>>> stn.station_info_gaps()
[{'rinex_count': 1, 'record_start': None, 'record_end': pyStationInfo.StationInfoRecord( BAQU                    2005 297 00 00 00  2005 301 00 49 01   0.0460  DHARP   0.0000   0.0000  TRIMBLE 4000SSE       NP 7.19 / S            7.19  2000                  TRM14532.00      NONE   66306               )}]
>>> 

but are shown in green in the web front:

[image]

Salacas, 2024-08-09:
Done.
Commit 0b4402b solved the issue.

marafigue, 2024-08-13:
This is not working correctly
[image]

Salacas, 2024-08-14:
Did you correct the gap already? Because I'm trying to figure out what rinex is causing the problem (between stationinfo 2007-2008) in the database and none of them seem to be out of stationinfo date range.

marafigue, 2024-08-14:
Yes, I fixed it. But this happened with several stations I was checking yesterday. I can find another example if you need it.

Salacas, 2024-08-14:
Yes, please. If you find one, leave the gap open so I can check

marafigue, 2024-08-14:
Station sag.live has a different problem, but still the icon is green:
[image]

Salacas, 2024-08-14:
The app isn't handling that kind of problem. We were asked to show the red icon when there are rinex gaps and when there isn't any stationinfo associated with the station. Could you provide another example where the problem is that there are gaps in the station?

marafigue, 2024-08-14:
I just deleted one row of the station info of station sag.bsar (that I needed to change completely anyways). I refreshed the site several times and waited for 10 minutes, but still the icon is not reflecting the problem:
[image]

marafigue, 2024-08-14:
On the other hand, if the app is not handling the "unknown antenna" problem, there is something wrong with the red icons too. In this case, for station sag.bsjl, it is showing that the station has problems, but the only problems are the "unknown antenna" type:
[image]

Salacas, 2024-08-14:
Solved. The process which sets the station state wasn't running. By rebooting the docker container, the process now is running every minute (based on actual conf).

closed 2024-08-14 by Salacas

demiangomez, 2024-08-15:
> On the other hand, if the app is not handling the "unknown antenna" problem, there is something wrong with the red icons too. In this case, for station sag.bsjl, it is showing that the station has problems, but the only problems are of the "unknown antenna" type:
> [image]
> 

Please note that the requested implementation did not include the "unknown antenna" issue. So the icon will be green for those sites even if Integrity Check says it has a problem. This is because this check cannot be done without the ANTEX file, which is currently not accessible from the front or backend.