demiangomez/geode #120: Is checking for regional vs global still needed?

demiangomez, 2024-10-17:
It appears that we are still checking for two different solution types

https://github.com/demiangomez/Parallel.GAMIT/blob/99a7a4dc85ce9dd34af5bbe716733ef0a00c6205/pgamit/pyNetwork.py#L167

but the calls to make_clusters is the same. If the regional branch is not needed anymore, please remove it or duplicate the if statement that checks for the number of stations (the the else branch of the above referenced if). We just run a test and it failed:

 >> 2024-10-17 16:37:38 g2regain 2024 268 -> Creating network clusters
 --  Processing type is regional with 7 active stations
Traceback (most recent call last):
  File "/fs/project/gomez.124/opt/conda/pgamit/bin/ParallelGamit.py", line 756, in <module>
    main()
  File "/fs/project/gomez.124/opt/conda/pgamit/bin/ParallelGamit.py", line 399, in main
    sessions = ExecuteGamit(cnn, JobServer, GamitConfig, stations, check_stations, args.ignore_missing, dates,
  File "/fs/project/gomez.124/opt/conda/pgamit/bin/ParallelGamit.py", line 700, in ExecuteGamit
    net_object = Network(cnn, archive, GamitConfig, stations, date, check_stations, ignore_missing)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/site-packages/pgamit/pyNetwork.py", line 170, in __init__
    clusters, ties = self.make_clusters(stations.get_active_coordinates(date), stn_active)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/site-packages/pgamit/pyNetwork.py", line 208, in make_clusters
    OC = over_cluster(qmean.labels_, points, metric='euclidean',
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/site-packages/pgamit/NetClusters.py", line 174, in over_cluster
    D, _ = nbrs.kneighbors(coordinates[nonmembers, :])
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/site-packages/sklearn/neighbors/_base.py", line 825, in kneighbors
    X = self._validate_data(X, accept_sparse="csr", reset=False, order="C")
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/site-packages/sklearn/base.py", line 633, in _validate_data
    out = check_array(X, input_name="X", **check_params)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/site-packages/sklearn/utils/validation.py", line 1087, in check_array
    raise ValueError(
ValueError: Found array with 0 sample(s) (shape=(0, 3)) while a minimum of 1 is required by NearestNeighbors.

the reason is because this day only has 7 active stations (it is a recent day, not all RINEX available yet).

espg, 2024-10-17:
do the changes in #121 address this sufficiently to merge #118 ?

espg, 2024-10-18:
fixed and merged in above PRs

closed 2024-10-18 by espg