demiangomez/geode #123: Strange issue in PGAMIT run

demiangomez, 2024-10-17:
@espg any ideas of what could be going on here?

>> 2024-10-17 19:51:42 Creating GAMIT session instances and executing GAMIT, please wait...
 >> 2024-10-17 19:51:42 g2global 2024 041 -> Creating network clusters
 --  Processing type is global with 731 active stations
Traceback (most recent call last):
  File "/fs/project/gomez.124/opt/conda/pgamit/bin/ParallelGamit.py", line 756, in <module>
    main()
  File "/fs/project/gomez.124/opt/conda/pgamit/bin/ParallelGamit.py", line 399, in main
    sessions = ExecuteGamit(cnn, JobServer, GamitConfig, stations, check_stations, args.ignore_missing, dates,
  File "/fs/project/gomez.124/opt/conda/pgamit/bin/ParallelGamit.py", line 700, in ExecuteGamit
    net_object = Network(cnn, archive, GamitConfig, stations, date, check_stations, ignore_missing)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/site-packages/pgamit/pyNetwork.py", line 187, in _init_
    backbone = self.backbone_delauney(stations.get_active_coordinates(date), stn_active)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/site-packages/pgamit/pyNetwork.py", line 257, in backbone_delauney
    geocenter = copy.deepcopy(stations2[0])
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 172, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 271, in _reconstruct
    state = deepcopy(state, memo)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 146, in deepcopy
    y = copier(x, memo)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 231, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 172, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 271, in _reconstruct
    state = deepcopy(state, memo)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 146, in deepcopy
    y = copier(x, memo)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 231, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 172, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 271, in _reconstruct
    state = deepcopy(state, memo)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 146, in deepcopy
    y = copier(x, memo)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 231, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 146, in deepcopy
    y = copier(x, memo)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 206, in _deepcopy_list
    append(deepcopy(a, memo))
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/copy.py", line 161, in deepcopy
    rv = reductor(4)
TypeError: cannot pickle 'odict_values' object

demiangomez, 2024-10-18:
It seems this is related to psycopg2. It returns a odict_values type for the fields instead of a tuple. Will be addressed in the next push.

espg, 2024-10-18:
@demiangomez will you please rebase and merge #121 before pushing any changes related to `odict_values` into master? PR #121  currently passes all tests, so the fix for `odict_values` should happen on top of those commits.

closed 2024-10-21 by demiangomez