demiangomez/geode #161: list index out of range when in create_gamit_sessions

demiangomez, 2024-12-03:
### Describe the bug

When running ParallelGamit.py, the generation of subnets stalls with an error. See traceback. We called PGAMIT using only the Canadian stations.

### Steps/Code to Reproduce

ParallelGamit.py g2na.cfg -p -dp 2024 1-29 

### Expected Results

A PGAMIT run

### Actual Results

>> 2024-12-02 20:48:00 g2na 2024 001 -> Creating network clusters
 --  Processing type is regional with 102 active stations
Traceback (most recent call last):
  File "/fs/project/gomez.124/opt/conda/pgamit/bin/ParallelGamit.py", line 763, in <module>
    main()
  File "/fs/project/gomez.124/opt/conda/pgamit/bin/ParallelGamit.py", line 399, in main
    sessions = ExecuteGamit(cnn, JobServer, GamitConfig, stations, check_stations, args.ignore_missing, dates,
  File "/fs/project/gomez.124/opt/conda/pgamit/bin/ParallelGamit.py", line 707, in ExecuteGamit
    net_object = Network(cnn, archive, GamitConfig, stations, date, check_stations, ignore_missing)
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/site-packages/pgamit/network.py", line 177, in _init_
    self.sessions = self.create_gamit_sessions(cnn, archive, clusters,
  File "/fs/project/gomez.124/opt/conda/pgamit/lib/python3.10/site-packages/pgamit/network.py", line 392, in create_gamit_sessions
    clusters['stations'][c], ties[c],
IndexError: list index out of range

### Versions

```shell
NA
```

espg, 2024-12-03:
I'll need the station data to debug this. Pandas [dataframe](https://pandas.pydata.org/docs/user_guide/index.html) would be ideal (see #162 ), but csv is fine too. 

also, when I finish the individual cluster plotting code, I'll have it read from a dataframe too.

demiangomez, 2024-12-03:
CSV file with stations in Canada

[can_stations_20241203.csv](https://github.com/user-attachments/files/17998869/can_stations_20241203.csv)

espg, 2024-12-03:
@demiangomez thank you, having a look

demiangomez, 2024-12-04:
Did you figure it out? We are stalled currently. Thanks.

espg, 2024-12-04:
yeah, I know what the issue is, just trying to decide how to solve it.

The issue is here: https://github.com/demiangomez/Parallel.GAMIT/blob/ac471fd71890a9e7363de97f97d4d80bc08390e6/pgamit/network.py#L387

That line is calculating the number of clusters using the centroids array length. It's an arbitrary choice, or at least it was when both the `centroids` and `clusters` data dictionaries / lists / objects were the same length. They're not anymore--

https://github.com/demiangomez/Parallel.GAMIT/blob/ac471fd71890a9e7363de97f97d4d80bc08390e6/pgamit/network.py#L185

this line above happens before the new pruning call, here:

https://github.com/demiangomez/Parallel.GAMIT/blob/ac471fd71890a9e7363de97f97d4d80bc08390e6/pgamit/network.py#L191

That line is by design removing clusters...but just to the `clusters` data object, not for the `centroids` (these should be in the same data structure so they update together, not as separate dictions in a list collection). This also means that the plot function is still plotting referencing the 'unpruned' data, which isn't what we want either. That's also why just switching line 387 to calculate length using the `clusters` object won't work; it'll fix the indexing error, but their would be buggy behavior since the centroids are no longer guaranteed to match to the correct corresponding paired cluster.

I started modifying the pruning function to return an index for filtering instead of a compressed boolean array, and reordering the centriod calculation to happen after the pruning... but I don't really like that approach. The code in that block is getting messy, ad hoc, and hard to maintain. Most / all of the errors that have happened with the clustering development are related to these ad hoc data structures that aren't documented or defined, so writing a fix there feels like another bandaid on top of a pile of bandaids.

A longer term solution I'm looking at now instead is masking/marking the pruned clusters inside of pandas array using NaNs that gets passed gamitsession, with exception behavior so that the dataframe will get written to disk when we catch an exception, to automate generation of debugging information for bug reports-- and which would include both pre-pruned clustering and post-pruned, incase we want to inspect that, which we likely would for the paper submission. Accessing the pruned data inside of pgamit would just be calling `drop_na` on the dataframe, and then the write to postgresql can also be handled by pandas. The big idea is that the whole dataframe object is what's modified, so everything is kept in sync, documented, and accessible for inspection as needed. 

Anyway, I'll roll out a PR to play with this before the end of the week and give some more details on what it does there.

espg, 2024-12-05:
I'll get you a temporary fix tonight so that it doesn't block or delay the clustering timing runs. 

Also happy to setup a testing branch for pandas integration and help you work on the longer term data structure solution. Ideally this is something we can get it up and running before the paper submission, although if it's not quite ready by then we can also fold it into the manuscript during reviewer revisions. I'm applying to this [open science call](https://new.nsf.gov/funding/opportunities/geo-ose-geosciences-open-science-ecosystem/nsf25-506/solicitation) in February, so I will need evidence of ongoing open science efforts to make the submission credible to nsf.

demiangomez, 2024-12-06:
There is another problem now: there cannot be empty networks. The process is generating empty networks which is creating GAMIT runs without stations:

2024-12-06 11:25:18 -> GamitTask initialized for g2na.OSU001 2024 001 local folder production/gamit/2024/001/g2na.OSU001 solution folder /fs/project/gomez.124/gamit_soln_osu/2024/001/g2na.OSU001
2024-12-06 11:25:18 -> g2na.OSU001 2024 1 executing on u112.asc.ohio-state.edu systems G, E, R
2024-12-06 11:25:18 -> Copied structure from /fs/project/gomez.124/gamit_soln_osu/2024/001/g2na.OSU001 to production/gamit/2024/001/g2na.OSU001
2024-12-06 11:25:18 -> Pre-process tasks (symlink fix) finished with no errors
2024-12-06 11:25:18 -> fetching orbits
2024-12-06 11:25:18 -> sp3 orbit found: COD[0-9]OPSFIN_20240010000_01D_[0-1]5M_ORB.SP3 -> production/gamit/2024/001/g2na.OSU001/igs/cod22951.sp3
2024-12-06 11:25:18 -> renaming production/gamit/2024/001/g2na.OSU001/igs/cod22951.sp3 to production/gamit/2024/001/g2na.OSU001/igs/igs22951.sp3
2024-12-06 11:25:18 -> fetching broadcast orbits
2024-12-06 11:25:18 -> broadcast orbit found: brdc0010.24n
2024-12-06 11:25:18 -> Ionex file found: /fs/project/gomez.124/orbits/ionex/2024/IGS0OPSFIN_20240010000_01D_02H_GIM.INX.gz -> production/gamit/2024/001/g2na.OSU001/igsg0010.24i.Z
2024-12-06 11:25:18 -> About to execute GAMIT - system count G: 0 E: 0 R: 0
2024-12-06 11:25:18 -> Not enough observing stations for system G. Skipping processing.
2024-12-06 11:25:18 -> Not enough observing stations for system E. Skipping processing.
2024-12-06 11:25:18 -> Not enough observing stations for system R. Skipping processing.
2024-12-06 11:25:18 -> No valid system solutions. Check fatals.
2024-12-06 11:25:19 -> Copy results from production/gamit/2024/001/g2na.OSU001 to /fs/project/gomez.124/gamit_soln_osu/2024/001/g2na.OSU001

The behavior of the clustering algorithm should be the same as before. Each cluster NEEDS to have stations. If the pruning algorithm removes one subnet, then there should not be an empty list in clusters and cluster_ties.

espg, 2024-12-06:
ugh, what a dumb error. Here's what's happening-- see these lines:

https://github.com/demiangomez/Parallel.GAMIT/blob/6717b18afbe1d596be9c8f32f883c045d3cbeb85/pgamit/cluster.py#L41-L46

`counts` is tally of how often a single station is represented across the clusters; if the `counts` for any station is under one, we have a problem. A few lines above I've temporarily removed the current cluster entirely:

https://github.com/demiangomez/Parallel.GAMIT/blob/6717b18afbe1d596be9c8f32f883c045d3cbeb85/pgamit/cluster.py#L36-L41

...so if removing a cluster causes zero 'problems', we zero out the row and add the cluster index to the `subset` list to filter out on line 46, which to my eyes is pretty clear:

https://github.com/demiangomez/Parallel.GAMIT/blob/6717b18afbe1d596be9c8f32f883c045d3cbeb85/pgamit/cluster.py#L46

We're casting the `subset` list to an array so we can index and filter these clusters. But the `subset` list/array isn't a listing of the clusters that we want-- it's a list/array of the clusters ***that we don't want***. So that's why there's a `~` in front of them, because that's the negation symbol in python; i.e., if you have a boolean array of `x = np.array([0, 1, 1, 0, 1])` , and you call `~x`, you get back the negation, which is `[1, 0, 0, 1, 0]`. So line 46 pretty clearly says "give all indices that aren't in `subset`", which is the standard way of doing this in a dataframe.

So what's happening?

Well, apparently numpy is screwing up:
```
>>> subset
[0, 10, 15, 19, 21, 22, 25, 31, 32, 46, 47, 67, 74, 77, 83]
>>> np.shape(OC)
(90, 726)
>>> np.shape(OC[~np.array(subset)])
(15, 726)
>>> np.shape(OC[np.array(subset)])
(15, 726)
```

There are 15 clusters out of 90 to be pruned. So `np.shape(OC[~np.array(subset)])` should be `(75, 726)`, but it's `(15, 726)`, which is the same as if we use don't use negation and just ask for the 15 row indices. You might be temped to think that numpy is just ignoring the tilde negation, but that's not true either:

```
>>> np.alltrue(OC[~np.array(subset)] == OC[np.array(subset)])
False
>>> np.sum(~(OC[~np.array(subset)] == OC[np.array(subset)]))
201
```

I genuinely have no idea what numpy is doing, nor do I care to find out. Remember what I said about a bandaid on top of other bandaids? Tilda negation is valid syntax for dataframes, [and is supported in pandas already](https://discovery.cs.illinois.edu/guides/DataFrame-Row-Selection/dataframe-isin-selection/), so the solution is to install pandas and use that. This is why pandas exist, to augment functionality that numpy does poorly or not at all. It will take me less time to adapt this with pandas than for me to come up and test yet another ad hoc work around.

I want to make clear that even though pandas understands the `~` negation operator, that's not even how this problem would have been solved. Instead, line 40 would (starting with PR #160 ) have been using NaNs, and then I'd just call `OC.dropna()` or `df.dropna()` which would make this code significantly shorter and error free.

This is what I'm going to do fix this. First, I'm going finish up the PR I'm working on to get the automatic pgamit installation and versioning setup with the build system and pip installation-- and while I do that, I'm going to add pandas as a dependency. Then I'll push out a fix for this using pandas. My guess is that I will have the install PR up on github next week. 

@demiangomez I recommend that if you haven't already done the clustering runs *without* using pruning, that you do that now and send me along the benchmarks when they finish next week. Then you can rerun them with pruning activated (once I have a fix here) and we can incorporate that measured difference as part of the paper discussion. You can run things on the existing code by reading and applying this comment:

https://github.com/demiangomez/Parallel.GAMIT/blob/6717b18afbe1d596be9c8f32f883c045d3cbeb85/pgamit/network.py#L189-L190

espg, 2025-02-25:
This is fixed by #189 ; below is our previous behavior:

```python
 >>> subset
[0, 10, 15, 19, 21, 22, 25, 31, 32, 46, 47, 67, 74, 77, 83]
>>> np.shape(OC)
(90, 726)
>>> np.shape(OC[~np.array(subset)])
(15, 726)
>>> np.shape(OC[np.array(subset)])
(15, 726)
```

This is what was added:

```python
        # Cast subset list to pandas index
        dfIndex = pd.Index(subset)
        # Cast OC to pandas dataframe
        dfOC = pd.DataFrame(OC)
        # Apply the 'inverse' index; pruned is boolean numpy index array
        pruned = ~dfOC.index.isin(dfIndex)
        return OC[pruned], central_points[~np.array(pruned)]
```

This is the new behavior:

```python
>>> dfIndex = pd.Index(subset)
>>> dfIndex
Index([0, 10, 15, 19, 21, 22, 25, 31, 32, 46, 47, 67, 74, 77, 83], dtype='int64')
>>> dfOC = pd.DataFrame(OC)
>>> dfOC.index.isin(dfIndex) # regular subset index
array([ True, False, False, False, False, False, False, False, False,
       False,  True, False, False, False, False,  True, False, False,
       False,  True, False,  True,  True, False, False,  True, False,
       False, False, False, False,  True,  True, False, False, False,
       False, False, False, False, False, False, False, False, False,
       False,  True,  True, False, False, False, False, False, False,
       False, False, False, False, False, False, False, False, False,
       False, False, False, False,  True, False, False, False, False,
       False, False,  True, False, False,  True, False, False, False,
       False, False,  True, False, False, False, False, False, False])
>>> ~dfOC.index.isin(dfIndex) # flipped index, i.e., what we use
array([False,  True,  True,  True,  True,  True,  True,  True,  True,
        True, False,  True,  True,  True,  True, False,  True,  True,
        True, False,  True, False, False,  True,  True, False,  True,
        True,  True,  True,  True, False, False,  True,  True,  True,
        True,  True,  True,  True,  True,  True,  True,  True,  True,
        True, False, False,  True,  True,  True,  True,  True,  True,
        True,  True,  True,  True,  True,  True,  True,  True,  True,
        True,  True,  True,  True, False,  True,  True,  True,  True,
        True,  True, False,  True,  True, False,  True,  True,  True,
        True,  True, False,  True,  True,  True,  True,  True,  True])
>>> np.shape(OC[~dfOC.index.isin(dfIndex)]) # applying to our return arrays
(75, 726)
```

I've applied using pandas here as a minimal fix so we can get our runs setup ASAP... however, at some point, we should propagate dataframe usage as a more general data structure within pgamit.

closed 2025-02-25 by espg