demiangomez/geode #167: Slow response over VPN

demiangomez, 2024-12-07:
I think it has something to do with how many images there are, but for no good reason, when I navigate to a station, and request metadata information (or RINEX, etc), it takes a very long time for the system to respond. This delay persists back on the home page, even after leaving the station. It only starts responding well again after navigating to another station.

[image]

demiangomez, 2024-12-08:
I did some experiments, and I am pretty sure that the problem is related to loading images for the PDF. The size and quantity of the images makes the system respond very slowly over the VPN or on low bandwidth connections. My intention is to tackle this issue in the next round of improvements, but I think I would like to get rid of this problem now to facilitate loading data. Maybe we can put the PDF generation somewhere else so that it does not slow down opening the station page. I'm even willing to just disable the PDF option until we find a definite solution, since we are currently not using it.

Salacas, 2024-12-13:
Resolved by reducing the resolution of image previews and loading metadata file only.

closed 2024-12-13 by Salacas