demiangomez/geode #179: Selecting only END DATE for time series produces wrong output

demiangomez, 2025-01-07:
The time series is not plotted correctly when only the end date is provided
[image]

time series:
[image]

expected output is that the end date of the time series is 01/01/2020. This works correctly in pyETM, I think the problem is what is the filler for the start date.

demiangomez, 2025-01-07:
Need to modify code here https://github.com/demiangomez/Parallel.GAMIT/blob/6717b18afbe1d596be9c8f32f883c045d3cbeb85/pgamit/pyETM.py#L2266 to account for a tuple with [None, Date] so that the API can pass this when there is no begin date.

francolau, 2025-04-03:
@Salacas Check

closed 2026-09-29 by demiangomez